chipsalliance/silicon-notebooks #7: libmamba Failed to download package

Kuo-TingKai, 2022-07-20:
When I ran the second code cell the example notebook digital-inverter-openlane.ipynb in colab, 
```python=
import os
import pathlib
import sys

!curl -Ls https://micro.mamba.pm/api/micromamba/linux-64/latest | tar -xvj bin/micromamba
conda_prefix_path = pathlib.Path('conda-env')
site_package_path = conda_prefix_path / 'lib/python3.7/site-packages'
sys.path.append(str(site_package_path.resolve()))
CONDA_PREFIX = str(conda_prefix_path.resolve())
PATH = os.environ['PATH']
LD_LIBRARY_PATH = os.environ.get('LD_LIBRARY_PATH', '')
%env CONDA_PREFIX={CONDA_PREFIX}
%env PATH={CONDA_PREFIX}/bin:{PATH}
%env LD_LIBRARY_PATH={CONDA_PREFIX}/lib:{LD_LIBRARY_PATH}
!bin/micromamba create --yes --prefix $CONDA_PREFIX
!echo 'python ==3.7*' >> {CONDA_PREFIX}/conda-meta/pinned
!bin/micromamba install --yes --prefix $CONDA_PREFIX \
                        --channel litex-hub \
                        --channel main \
                        open_pdks.sky130a \
                        magic \
                        openroad \
                        netgen \
                        yosys 
!bin/micromamba install --yes --prefix $CONDA_PREFIX \
                        --channel conda-forge \
                        tcllib gdstk pyyaml click
```

The following error occurred:
```bash=
Extracting         0%error    libmamba Failed to download package from https://conda.anaconda.org/main/linux-64/_libgcc_mutex-0.1-main.tar.bz2 (status 403)
```

I think there was something wrong with the first micromamba install command.

proppy, 2022-09-07:
@Kuo-TingKai can you try again? I suspect this was a transient error.

Kuo-TingKai, 2022-10-28:
Ok I'll try later