chipsalliance/silicon-notebooks #13: cffi version mismatch

t-sasatani, 2022-09-05:
The following error appeared when running `simulator = circuit.simulator()` in the last cell, and I needed to run `!conda install cffi==1.15.1` to update it.

```
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
<ipython-input-6-496ab6cd4415> in <module>
     15                            pulse_width=1@u_ns, period=2@u_ns, delay_time=1@u_ns)
     16 print(str(circuit))
---> 17 simulator = circuit.simulator()
     18 analysis = simulator.transient(step_time=10@u_ps, end_time=2@u_ns)
     19 print('done')

4 frames
/content/conda-env/lib/python3.7/site-packages/PySpice/Spice/Netlist.py in simulator(self, *args, **kwargs)
   1290 
   1291     def simulator(self, *args, **kwargs):
-> 1292         return CircuitSimulator.factory(self, *args, **kwargs)

/content/conda-env/lib/python3.7/site-packages/PySpice/Spice/Simulation.py in factory(cls, circuit, *args, **kwargs)
   1156                 sub_cls = NgSpiceSubprocessCircuitSimulator
   1157             elif simulator == 'ngspice-shared':
-> 1158                 from .NgSpice.Simulation import NgSpiceSharedCircuitSimulator
   1159                 sub_cls = NgSpiceSharedCircuitSimulator
   1160         elif simulator in ('xyce-serial', 'xyce-parallel'):

/content/conda-env/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Simulation.py in <module>
     30 from ..Simulation import CircuitSimulator
     31 from .Server import SpiceServer
---> 32 from .Shared import NgSpiceShared
     33 
     34 ####################################################################################################

/content/conda-env/lib/python3.7/site-packages/PySpice/Spice/NgSpice/Shared.py in <module>
    108 ####################################################################################################
    109 
--> 110 ffi = FFI()
    111 
    112 ####################################################################################################

/usr/local/lib/python3.7/dist-packages/cffi/api.py in __init__(self, backend)
     54                     raise Exception("Version mismatch: this is the 'cffi' package version %s, located in %r.  When we import the top-level '_cffi_backend' extension module, we get version %s, located in %r.  The two versions should be equal; check your installation." % (
     55                         __version__, __file__,
---> 56                         backend.__version__, backend.__file__))
     57                 else:
     58                     # PyPy

Exception: Version mismatch: this is the 'cffi' package version 1.15.1, located in '/usr/local/lib/python3.7/dist-packages/cffi/api.py'.  When we import the top-level '_cffi_backend' extension module, we get version 1.15.0, located in '/content/conda-env/lib/python3.7/site-packages/_cffi_backend.cpython-37m-x86_64-linux-gnu.so'.  The two versions should be equal; check your installation.
```

proppy, 2022-09-07:
was this with the analog-inverter-magic notebook?
https://github.com/chipsalliance/silicon-notebooks/blob/main/analog-inverter-magic.ipynb

t-sasatani, 2022-09-08:
Yes, it happened with the analog-inverter-magic notebook.

closed 2022-09-12 by proppy