chipsalliance/silicon-notebooks #52: notebooks fail with library version issues

proppy, 2023-01-26:
```
/bin/bash: /content/conda-env/lib/libtinfo.so.6: no version information available (required by /bin/bash)
Cloning into 'subservient'...
/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libp11-kit.so.0: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0
/bin/bash: /content/conda-env/lib/libtinfo.so.6: no version information available (required by /bin/bash)
Cloning into 'serv'...
/usr/lib/git-core/git-remote-https: symbol lookup error: /lib/x86_64-linux-gnu/libp11-kit.so.0: undefined symbol: ffi_type_pointer, version LIBFFI_BASE_7.0
```

proppy, 2023-01-26:
This must be due to us messing with the `LD_LIBRARY_PATH` in the conda setup.

```
LD_LIBRARY_PATH=/content/conda-env/lib:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
```

closed 2023-01-26 by mithro

prarabdhshukla, 2023-05-09:
Omg @proppy! You're a lifesaver. I was just so frustrated until I saw this. This particular error cost me around 4 hours of precious time. I wish all these libraries had better documentation. As software grows, dependencies need to be managed more carefully, because a small tweak for a particular lib's dev, can be a million head scratching seconds for others!

PR #53 (2023-01-26): ipynb: fix library path
Fixes #52