chipsalliance/silicon-notebooks #59: digital inverter notebook does not work in Colab ([ERROR]: Flow failed.)

mu2519, 2023-03-05:
When I executed `digital-inverter-openlane.ipynb` in Colab, the following error occurred:
```
env: PDK=sky130A
OpenLane 2023.03.01_0_ge10820ec-conda
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'config.json'...
[INFO]: PDK Root: /content/conda-env/share/pdk
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /content/runs/RUN_2023.03.05_10.06.04
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[STEP 1]
[INFO]: Running Synthesis (log: runs/RUN_2023.03.05_10.06.04/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: runs/RUN_2023.03.05_10.06.04/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: runs/RUN_2023.03.05_10.06.04/logs/floorplan/3-initial_fp.log)...
[INFO]: Floorplanned with width 23.46 and height 35.36.
[STEP 4]
[INFO]: Running IO Placement...
[STEP 5]
[INFO]: Running Tap/Decap Insertion (log: runs/RUN_2023.03.05_10.06.04/logs/floorplan/5-tap.log)...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 6]
[INFO]: Generating PDN (log: runs/RUN_2023.03.05_10.06.04/logs/floorplan/6-pdn.log)...
[STEP 7]
[INFO]: Performing Random Global Placement (log: runs/RUN_2023.03.05_10.06.04/logs/placement/7-global.log)...
[STEP 8]
[INFO]: Running Placement Resizer Design Optimizations (log: runs/RUN_2023.03.05_10.06.04/logs/placement/8-resizer.log)...
[STEP 9]
[INFO]: Running Detailed Placement (log: runs/RUN_2023.03.05_10.06.04/logs/placement/9-detailed.log)...
[STEP 10]
[INFO]: Running Placement Resizer Timing Optimizations (log: runs/RUN_2023.03.05_10.06.04/logs/cts/10-resizer.log)...
[STEP 11]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: runs/RUN_2023.03.05_10.06.04/logs/routing/11-resizer.log)...
[STEP 12]
[INFO]: Running Detailed Placement (log: runs/RUN_2023.03.05_10.06.04/logs/routing/12-diode_legalization.log)...
[STEP 13]
[INFO]: Running Global Routing (log: runs/RUN_2023.03.05_10.06.04/logs/routing/13-global.log)...
[INFO]: Starting OpenROAD Antenna Repair Iterations...
[STEP 14]
[INFO]: Writing Verilog (log: runs/RUN_2023.03.05_10.06.04/logs/routing/13-global_write_netlist.log)...
[STEP 15]
[INFO]: Running Fill Insertion (log: runs/RUN_2023.03.05_10.06.04/logs/routing/15-fill.log)...
[STEP 16]
[INFO]: Running Detailed Routing (log: runs/RUN_2023.03.05_10.06.04/logs/routing/16-detailed.log)...
[INFO]: No DRC violations after detailed routing.
[STEP 17]
[INFO]: Checking Wire Lengths (log: runs/RUN_2023.03.05_10.06.04/logs/routing/17-wire_lengths.log)...
[STEP 18]
[INFO]: Running SPEF Extraction at the min process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/18-parasitics_extraction.min.log)...
[STEP 19]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/19-rcx_mcsta.min.log)...
[STEP 20]
[INFO]: Running SPEF Extraction at the max process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/20-parasitics_extraction.max.log)...
[STEP 21]
[INFO]: Running Multi-Corner Static Timing Analysis at the max process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/21-rcx_mcsta.max.log)...
[STEP 22]
[INFO]: Running SPEF Extraction at the nom process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/22-parasitics_extraction.nom.log)...
[STEP 23]
[INFO]: Running Multi-Corner Static Timing Analysis at the nom process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/23-rcx_mcsta.nom.log)...
[STEP 24]
[INFO]: Running Single-Corner Static Timing Analysis at the nom process corner (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/24-rcx_sta.log)...
[STEP 25]
[INFO]: Creating IR Drop Report (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/25-irdrop.log)...
[STEP 26]
[INFO]: Running Magic to generate various views...
[INFO]: Streaming out GDSII with Magic (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/26-gdsii.log)...
[INFO]: Generating MAGLEF views...
[STEP 27]
[INFO]: Streaming out GDSII with KLayout (log: runs/RUN_2023.03.05_10.06.04/logs/signoff/27-gdsii-klayout.log)...
[ERROR]: during executing: "python3 /content/conda-env/share/openlane/scripts/klayout/stream_out.py --output /content/runs/RUN_2023.03.05_10.06.04/results/signoff/inverter.klayout.gds --tech-file /content/conda-env/share/pdk/sky130A/libs.tech/klayout/tech/sky130A.lyt --def-layer-map-file /content/conda-env/share/pdk/sky130A/libs.tech/klayout/tech/sky130A.map --props-file /content/conda-env/share/pdk/sky130A/libs.tech/klayout/tech/sky130A.lyp --top inverter --with-gds-file /content/conda-env/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/gds/sky130_fd_sc_hd.gds --input-lef /content/runs/RUN_2023.03.05_10.06.04/tmp/merged.nom.lef /content/runs/RUN_2023.03.05_10.06.04/results/routing/inverter.def |& tee /dev/null /content/runs/RUN_2023.03.05_10.06.04/logs/signoff/27-gdsii-klayout.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
Unable to open file: /content/conda-env/share/pdk/sky130A/libs.tech/klayout/tech/sky130A.map (errno=2) in Layout.read
child process exited abnormally

[INFO]: Saving current set of views in 'runs/RUN_2023.03.05_10.06.04/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.03.05_10.06.04/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.03.05_10.06.04/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

I suspect that the source of the problem is the change in the environment because the same version of the same notebook did work previously.

mithro, 2023-03-06:
FYI - @proppy

proppy, 2023-03-07:
Looks like this is due to https://github.com/RTimothyEdwards/open_pdks/commit/1683283a55ace46a24193f37b39cf271a5ab1808

You can workaround this by selecting a different version for `open_pdks_version` in the first cell:
```
1.0.397
```

proppy, 2023-03-07:
We should probably pin dependencies, and have a bot similar to https://github.com/regro/cf-scripts that bump package versions.