chipsalliance/silicon-notebooks #63: Analog Notebooks

ThomasPluck, 2023-03-17:
I would like it if there were more notebooks that include analog workflows that could be done in Jupyter notebooks.

It would be nice to introduce some tools in Analog FOSSi like:

- OpenVAF/HDL21 for Hardware Description
- ALIGN /Laygo2 for Layout Automation

Alongside a few tutorials to explain aspects of the analog design flow, particularly:

- Design/Simulate Cycle
- Layout Verification

proppy, 2023-03-22:
@ThomasPluck +1, we also discussed the following notebooks ideas with @msaligane, @bmurmann and @joamatab
- circuit capture (xschem, skidl, schemdraw)
- ngspice simulation see https://github.com/sscs-ose/sscs-ose-chipathon.github.io/pull/6
- gdsfactory/klayout based layout 
- post layout extraction and simulation
- integration with digital flow

proppy, 2023-03-23:
It would also be good to have notebooks that demonstrate mixed signal simulation with:
- https://yosyshq.readthedocs.io/projects/yosys/en/latest/cmd/write_spice.html
- http://opencircuitdesign.com/magic/commandref/extresist.html