chipsalliance/silicon-notebooks #69: OpenLane Notebooks: "::env(FP_PDN_LOWER_LAYER)": no such variable

ThomasPluck, 2023-06-17:
Hello,

I've been trying to use the notebooks, and I came across this strange error at Step 6 of the OpenLane flow:

```red
[INFO]: Reading ODB at '/content/runs/RUN_2023.06.17_19.13.55/tmp/floorplan/5-tapcell.odb'...
define_corners Typical
read_liberty -corner Typical /content/conda-env/share/pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Error: pdn_cfg.tcl, 92 can't read "::env(FP_PDN_LOWER_LAYER)": no such variable
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

I'm not sure if this belongs on OpenLane instead, but I've seen this error on both the XLS Adder and  Digital Inverter.

Thanks,

Thomas Pluck

proppy, 2023-07-09:
related https://github.com/The-OpenROAD-Project/OpenLane/issues/1877

proppy, 2023-07-09:
This seems to be fixed with:
```
openlane            2023.06.26_0_g3bc9d02d  20230606_125334       litex-hub      12MB
```

proppy, 2023-07-17:
Related https://github.com/hdl/conda-eda/issues/343