chipsalliance/silicon-notebooks #72: Flow fails in step 3

rvijgen, 2023-08-24:
It seems like this worked a few days ago, but now I get this error:

[STEP 3]
[INFO]: Running Initial Floorplanning (log: runs/RUN_2023.08.24_09.15.35/logs/floorplan/3-initial_fp.log)...
[ERROR]: during executing: "openroad -exit -no_init -python /content/conda-env/share/openlane/scripts/odbpy/defutil.py extract_core_dims --output-data /content/runs/RUN_2023.08.24_09.15.35/tmp/dimensions.txt --input-lef /content/runs/RUN_2023.08.24_09.15.35/tmp/merged.nom.lef /content/runs/RUN_2023.08.24_09.15.35/tmp/floorplan/3-initial_fp.def"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
  File "/content/conda-env/lib/python3.7/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/content/conda-env/lib/python3.7/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/content/conda-env/share/openlane/scripts/odbpy/defutil.py", line 36, in extract_core_dims
    reader = OdbReader(input_lef, input_def)
  File "/content/conda-env/share/openlane/scripts/odbpy/reader.py", line 37, in __init__
    odb.read_def(self.db, def_in)
  File "odb_py.py", line 9553, in read_def
TypeError: in method 'read_def', argument 1 of type 'odb::dbTech *'

[ERROR]: Step 3 (floorplan) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {try_exec openroad -exit -no_init -python /content/conda-env/share/openlane/scripts/odbpy/defutil.py extract_core_dims --output-data /content/runs/RUN_2023.08.24_09.15.35/tmp/dimensions.txt --input-lef /content/runs/RUN_2023.08.24_09.15.35/tmp/merged.nom.lef /content/runs/RUN_2023.08.24_09.15.35/tmp/floorplan/3-initial_fp.def} CALL extract_core_dims CALL init_floorplan CALL run_floorplan CALL {run_non_interactive_mode -design .}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "try_exec" line 17)
    invoked from within
"try_exec $::env(OPENROAD_BIN) -exit -no_init -python $::env(SCRIPTS_DIR)/odbpy/defutil.py extract_core_dims --output-data $out_tmp --input-lef $::env(..."
    (procedure "extract_core_dims" line 9)
    invoked from within
"extract_core_dims"
    (procedure "init_floorplan" line 60)
    invoked from within
"init_floorplan"
    (procedure "run_floorplan" line 7)
    invoked from within
"run_floorplan"} -errorline 1
[INFO]: Saving current set of views in 'runs/RUN_2023.08.24_09.15.35/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/RUN_2023.08.24_09.15.35/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/RUN_2023.08.24_09.15.35/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

isledge, 2023-08-24:
This is most likely a version mismatch issue between `openroad` and `openlane`, and, potentially `open_pdks`, due to recent changes.  We'll have to wait for someone to provide a list of working versions, since none of the previous suggestions work.

isledge, 2023-08-24:
Cross-posting from [https://github.com/The-OpenROAD-Project/OpenLane/issues/1953#issuecomment-1691581648](url):

> This turned out to be an easy fix.
>
> In `/content/conda-env/share/openlane/scripts/odbpy/reader.py` you will need to modify line 37 to be `odb.read_def(self.db.getTech(), def_in)`. Currently, line 37 reads `odb.read_def(self.db, def_in)`.

Lawrence-lugs, 2023-08-31:
Issue was fixed for the notebook we were using by fixing the versions to the following:

```python
#@title Install dependencies {display-mode: "form"}
#@markdown - Click the ▷ button to setup the digital design environment based on [conda-eda](https://github.com/hdl/conda-eda).

openlane_version = '2023.07.25_2_g01e67230' #@param {type:"string"}
openroad_version = '2.0_9484_g353633e01' #@param {type:"string"}
open_pdks_version = '1.0.428_2_g405227f' #@param {type:"string"}

if openlane_version == 'latest':
  openlane_version = ''
if open_pdks_version == 'latest':
  open_pdks_version = ''
if openroad_version == 'latest':
  openroad_version = ''

import os
import pathlib
!curl -Ls https://micro.mamba.pm/api/micromamba/linux-64/latest | tar -xj bin/micromamba
conda_prefix_path = pathlib.Path('conda-env')
CONDA_PREFIX = str(conda_prefix_path.resolve())
!bin/micromamba create --yes --prefix $CONDA_PREFIX
!echo 'python ==3.7*' >> {CONDA_PREFIX}/conda-meta/pinned
!CI=0 bin/micromamba install --yes --prefix $CONDA_PREFIX \
                     --channel litex-hub \
                     --channel main \
                     openlane={openlane_version} \
                     open_pdks.sky130a={open_pdks_version} \
                     openroad={openroad_version} \
                     iverilog
!python -m pip install gdstk gdstk
PATH = os.environ['PATH']
%env CONDA_PREFIX={CONDA_PREFIX}
%env PATH={CONDA_PREFIX}/bin:{PATH}
```